openbmc/linux #220: WARN at init/main.c:1061 start_kernel+0x4bc/0x680

shenki, 2023-02-27:
@legoater reported a warning with dev-6.1 on ast2500. I can reproduce on qemu with his configuration:

```
[    0.002218] ------------[ cut here ]------------
[    0.002338] WARNING: CPU: 0 PID: 0 at init/main.c:1061 start_kernel+0x4bc/0x680
[    0.002785] Interrupts were enabled early
[    0.002872] Modules linked in:
[    0.003077] CPU: 0 PID: 0 Comm: swapper Not tainted 6.1.12-00100-gc6467d425900 #13
[    0.003268] Hardware name: Generic DT based system
[    0.003399] Backtrace: 
[    0.003525]  dump_backtrace from show_stack+0x20/0x24
[    0.003704]  r7:80b01020 r6:00000009 r5:00000153 r4:80a12190
[    0.003822]  show_stack from dump_stack_lvl+0x2c/0x34
[    0.003921]  dump_stack_lvl from dump_stack+0x14/0x1c
[    0.004022]  r5:00000425 r4:809fd2b0
[    0.004083]  dump_stack from __warn+0x80/0xfc
[    0.004170]  __warn from warn_slowpath_fmt+0x90/0xb4
[    0.004268]  r7:80b01020 r6:00000425 r5:809fd2b0 r4:809fd2e8
[    0.004357]  warn_slowpath_fmt from start_kernel+0x4bc/0x680
[    0.004465]  r8:00000000 r7:ffffffff r6:00c0387d r5:9dfffd00 r4:80cb1000
[    0.004574]  start_kernel from 0x0
[    0.004719]  r10:00c5387d r9:410fb767 r8:881f8000 r7:ffffffff r6:00c0387d r5:00000051
[    0.004846]  r4:80b0033c
[    0.004917] ---[ end trace 0000000000000000 ]---
```

This can be reproduced by building aspeed_g5_defconfig with CONFIG_SMP=n and CONFIG_DEBUG_KMEMLEAK=y.

```
$ make ARCH=arm O=aspeed-g5-dev aspeed_g5_defconfig
$ ./scripts/config --file aspeed-g5-dev/.config --disable SMP 
$ ./scripts/config --file aspeed-g5-dev/.config --enable CONFIG_DEBUG_KMEMLEAK
```

shenki, 2023-02-28:
v5.17 does not warn.
v5.18 does.

(Note: when building pre-5.16 kernels with Debian's armhf cross compiler, cherry pick 418ace9992a7647c446ed3186df40cf165b67298)

Bisection points to b87cf9118e03d8a83039c88fa348165148f5d487 as the culprit.

```
$ git bisect log 
git bisect start
# status: waiting for both good and bad commits
# good: [f443e374ae131c168a065ea1748feac6b2e76613] Linux 5.17
git bisect good f443e374ae131c168a065ea1748feac6b2e76613
# status: waiting for bad commit, 1 good commit known
# bad: [4b0986a3613c92f4ec1bdc7f60ec66fea135991f] Linux 5.18
git bisect bad 4b0986a3613c92f4ec1bdc7f60ec66fea135991f
# bad: [25fd2d41b505d0640bdfe67aa77c549de2d3c18a] selftests: kselftest framework: provide "finished" helper
git bisect bad 25fd2d41b505d0640bdfe67aa77c549de2d3c18a
# bad: [b4bc93bd76d4da32600795cd323c971f00a2e788] Merge tag 'arm-drivers-5.18' of git://git.kernel.org/pub/scm/linux/kernel/git/soc/soc
git bisect bad b4bc93bd76d4da32600795cd323c971f00a2e788
# good: [3fe2f7446f1e029b220f7f650df6d138f91651f2] Merge tag 'sched-core-2022-03-22' of git://git.kernel.org/pub/scm/linux/kernel/git/tip/tip
git bisect good 3fe2f7446f1e029b220f7f650df6d138f91651f2
# good: [182966e1cd74ec0e326cd376de241803ee79741b] Merge tag 'media/v5.18-1' of git://git.kernel.org/pub/scm/linux/kernel/git/mchehab/linux-media
git bisect good 182966e1cd74ec0e326cd376de241803ee79741b
# good: [49a24e9d9c740d3bd8b1200f225f67d45e3d68a5] Make the SOF control, PCM and PM code IPC agnostic
git bisect good 49a24e9d9c740d3bd8b1200f225f67d45e3d68a5
# bad: [8ffa5709e577385a1c8d20fb434cb02732f1d991] Merge tag 'arm-defconfig-5.18' of git://git.kernel.org/pub/scm/linux/kernel/git/soc/soc
git bisect bad 8ffa5709e577385a1c8d20fb434cb02732f1d991
# good: [e6aef3496a00a12e78a571f61d98300cf0a86e6a] Merge tag 'm68knommu-for-v5.18' of git://git.kernel.org/pub/scm/linux/kernel/git/gerg/m68knommu
git bisect good e6aef3496a00a12e78a571f61d98300cf0a86e6a
# bad: [9c0e6a89b592f4c4e4d769dbc22d399ab0685159] Merge tag 'for-linus' of git://git.armlinux.org.uk/~rmk/linux-arm
git bisect bad 9c0e6a89b592f4c4e4d769dbc22d399ab0685159
# bad: [cafc0eab168917ec9c0cd47d530a40cd40eb2928] ARM: v7m: enable support for IRQ stacks
git bisect bad cafc0eab168917ec9c0cd47d530a40cd40eb2928
# good: [b832faec33d4e27c32490c0732098ecf79b222af] ARM: switch_to: clean up Thumb2 code path
git bisect good b832faec33d4e27c32490c0732098ecf79b222af
# good: [54f481a2308efab49d2b14c3f8263b34fdb1c65e] ARM: remove old-style irq entry
git bisect good 54f481a2308efab49d2b14c3f8263b34fdb1c65e
# good: [4e918ab13eaf40f19938659cb5a22c93172778a8] ARM: assembler: add optimized ldr/str macros to load variables from memory
git bisect good 4e918ab13eaf40f19938659cb5a22c93172778a8
# bad: [b87cf9118e03d8a83039c88fa348165148f5d487] ARM: use TLS register for 'current' on !SMP as well
git bisect bad b87cf9118e03d8a83039c88fa348165148f5d487
# good: [7b9896c352073156a325c3bb0dc4c46e06e2a468] ARM: percpu: add SMP_ON_UP support
git bisect good 7b9896c352073156a325c3bb0dc4c46e06e2a468
# first bad commit: [b87cf9118e03d8a83039c88fa348165148f5d487] ARM: use TLS register for 'current' on !SMP as well
```

mdmillerii, 2023-03-01:
Considering that https://github.com/openbmc/linux/commit/b87cf9118e03d8a83039c88fa348165148f5d487 is a kconfig change that implies CONFIG_SMP was already broken do you plan to bisect that config (I'd expect on UP qemu to be a sufficient test)?