openbmc/linux #221: Bug in USB gadget causes fail of more than one USB device connecting to AST2500 USB HVUB

Kitsok, 2023-03-05:
Hello!

I've found an issue with the USB gadget driver: if virtual media is connected to USB hub, then obmc-ikvm fails due to inability to set up USB HID device:
`Mar 03 19:20:31 bmc kernel: Error: Driver 'configfs-gadget' is already registered, aborting...
Mar 03 19:20:31 bmc kernel: UDC core: obmc_hid: driver registration failed: -16
Mar 03 19:20:31 bmc obmc-ikvm[455]: 03/03/2023 19:20:31   0 other clients
Mar 03 19:20:31 bmc obmc-ikvm[455]: Failed to connect HID gadget`

I've found that only one device can be connected despite the VHUB has 5 ports.

Further googling lead me to this topic: https://lore.kernel.org/lkml/20220907112210.11949-1-linhaoguo86@gmail.com/
Applying the patch fixed the problem.

thangqn-ampere, 2023-03-06:
This should be fixed on the latest openbmc which uses linux from dev-6.1 branch that includes https://github.com/openbmc/linux/commit/96abcf7230d7b860c445e2ea8645524c31d0106c.

shenki, 2023-03-06:
This should be resolved in dev-6.1. Please re-open if you find otherwise.

closed 2023-03-06 by shenki