openbmc/linux #233: Is the SSH URI of "git@github.com:openbmc/linux.git" in correct format?

zlu38111, 2025-01-29:
It looks the symbol of ":" is a typo of "/" because the "openbmc" following ":" will be parsed as a port number in BITBAKE tool.

Loaded 5157 entries from dependency cache.
ERROR: /hdd/work/openbmc/meta-aspeed/recipes-kernel/linux/linux-aspeed_git.bb: Error executing a python function in <code>:

The stack trace of python calls that resulted in this exception/failure was:
File: '<code>', lineno: 3, function: <module>
     0001:__anon_20__hdd_work_openbmc_meta_conf_machine_include_arm_feature_arm_thumb_inc(d)
     0002:__anon_34__hdd_work_openbmc_meta_classes_global_patch_bbclass(d)
 *** 0003:__anon_706__hdd_work_openbmc_meta_classes_global_base_bbclass(d)
     0004:__anon_1210__hdd_work_openbmc_meta_classes_global_insane_bbclass(d)
     0005:__anon_1609__hdd_work_openbmc_meta_classes_global_insane_bbclass(d)
     0006:__anon_243__hdd_work_openbmc_meta_classes_global_package_bbclass(d)
     0007:__anon_287__hdd_work_openbmc_meta_classes_global_package_ipk_bbclass(d)
File: '/hdd/work/openbmc/meta/classes-global/base.bbclass', lineno: 598, function: __anon_706__hdd_work_openbmc_meta_classes_global_base_bbclass
     0594:                    raise bb.parse.SkipRecipe("it has incompatible license(s): %s" % ' '.join(incompatible_lic))
     0595:
     0596:    srcuri = d.getVar('SRC_URI')
     0597:    for uri_string in srcuri.split():
 *** 0598:        uri = bb.fetch.URI(uri_string)
     0599:        # Also check downloadfilename as the URL path might not be useful for sniffing
     0600:        path = uri.params.get("downloadfilename", uri.path)
     0601:
     0602:        # HTTP/FTP use the wget fetcher
File: '/hdd/work/openbmc/poky/bitbake/lib/bb/fetch2/__init__.py', lineno: 254, function: __init__
     0250:            self.relative = True
     0251:
     0252:        if not self.relative:
     0253:            self.hostname = urlp.hostname or ''
 *** 0254:            self.port = urlp.port
     0255:
     0256:            self.userinfo += urlp.username or ''
     0257:
     0258:            if urlp.password:
File: '/usr/lib/python3.10/urllib/parse.py', lineno: 186, function: port
     0182:        if port is not None:
     0183:            if port.isdigit() and port.isascii():
     0184:                port = int(port)
     0185:            else:
 *** 0186:                raise ValueError(f"Port could not be cast to integer value as {port!r}")
     0187:            if not (0 <= port <= 65535):
     0188:                raise ValueError("Port out of range 0-65535")
     0189:        return port
     0190:
Exception: ValueError: Port could not be cast to integer value as 'openbmc'

ERROR: Parsing halted due to errors, see error messages above

sinuscosinustan, 2025-01-29:
The Git URL is usually defined in the `git://` format and uses HTTPS by default (see [linux-aspeed.inc](https://github.com/openbmc/openbmc/blob/master/meta-aspeed/recipes-kernel/linux/linux-aspeed.inc#L9)).

Do you have an override in your Git configuration, or did you make a change to `linux-aspeed`?

Besides that, this issue should be opened in `openbmc/openbmc` and not here, as it is not related to the Linux source code.

williamspatrick, 2025-01-29:
Please give more context on what you are seeing.  The `SRC_URI` you mentioned looks nothing like what we have in the openbmc tree.

```
meta-aspeed/recipes-kernel/linux/linux-aspeed.inc
9:KSRC ?= "git://github.com/openbmc/linux;protocol=https;branch=${KBRANCH}"
```

closed 2025-01-29 by williamspatrick

zlu38111, 2025-01-29:
the URI of "github.com/openbmc/linux" is not existing,  the real URI of
Linux-aspeed is "github.com:openbmc/linx".

I tried to build the latest openbmc many times and found only this URI is
not existing, others are working fine.  So I guess probably the ":" is a.
typo of "/".
Then I tried to do  "git clone ***@***.***:openbmc/linux", and it
worked, but "git clone ***@***.***/openbmc/linx" did not work.  But
after replacing the "github.com/openbmc/linux" with
"github.com:openbmc/linux",
the bitbake failed with parsing the URL and took the ":openbmc" as a port
number (see the logs in my last report).

zlu38111, 2025-01-29:
Yes, I change the URI for test.  I did multiple tests to verify the root
cause.  Only there are multiple URIs in the format but only this URI of
"linux-aspeed" does not work, and finally found the URI "
github.com/openbmc/linux" not existing.  And even, the command of "git
clone ***@***.***/openbmc/linux" did not work,  but "git clone.
***@***.***:openbmc/linux" worked.

So I guess maybe there is a typo in the github URI, not openbmc.

Thanks,

williamspatrick, 2025-01-29:
What code are you referring to?  I don't see any colon in this: https://github.com/openbmc/openbmc/blob/c046321bd260a1a542a78b6215ef6c6068fd2a68/meta-aspeed/recipes-kernel/linux/linux-aspeed.inc#L9

zlu38111, 2025-01-29:
See attached screenshot.

williamspatrick, 2025-01-29:
> See attached screenshot.

Screenshot did not get attached.

williamspatrick, 2025-01-29:
> That's weird.  I can see it in  my sent mail.
> 
> Step 1:  go to link:  https://github.com/openbmc/linux
> step  2:  click the "Code" tab, and "SSH" tab.
>     you can see the URI ***@***.***:openbmc/linux.git".
> […](#)

You're opening a task to us because you don't like how Github shows URLs in all their repositories?  This is the same everywhere.  Every single repository in Github has the exact same format.

Originally you mentioned bitbake issues, which made us all think you were talking about an issue in one of our recipes.  This means you're trying to write your own recipe?  Please use the ones we have in openbmc/openbmc.

williamspatrick, 2025-01-29:
BTW, that format (git@github.com:openbmc/linux.git) is usable by `git clone`, which is what Github is probably aiming for.

zlu38111, 2025-01-29:
> BTW, that format ***@***.***:openbmc/linux.git) is usable by git clone,

Thanks for your answer.
I built the "poky" without problem,  but
when tried to build "openbmc"  and got the fetch URI failed, like
==============================================================
WARNING: linux-aspeed-6.6.71+git-r0 do_fetch: Failed to fetch URL git://
github.com/openbmc/linux;protocol=https;branch=dev-6.6, attempting MIRRORS
if available
ERROR: linux-aspeed-6.6.71+git-r0 do_fetch: Fetcher failure: Fetch command
export PSEUDO_DISABLED=1; export
DBUS_SESSION_BUS_ADDRESS="unix:abstract=/tmp/dbus-R7pb6xzOs2,guid=492b8758dbe081da38f1ee306799b68a";
export
PATH="/hdd/work/openbmc/scripts:/hdd/work/openbmc/build/romulus/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/6.6.71+git/recipe-sysroot-native/usr/bin/arm-openbmc-linux-gnueabi:/hdd/work/openbmc/build/romulus/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/6.6.71+git/recipe-sysroot/usr/bin/crossscripts:/hdd/work/openbmc/build/romulus/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/6.6.71+git/recipe-sysroot-native/usr/sbin:/hdd/work/openbmc/build/romulus/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/6.6.71+git/recipe-sysroot-native/usr/bin:/hdd/work/openbmc/build/romulus/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/6.6.71+git/recipe-sysroot-native/sbin:/hdd/work/openbmc/build/romulus/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/6.6.71+git/recipe-sysroot-native/bin:/hdd/work/openbmc/poky/bitbake/bin:/hdd/work/openbmc/build/romulus/tmp/hosttools";
export HOME="/home/zlu"; LANG=C git -c gc.autoDetach=false -c
core.pager=cat -c safe.bareRepository=all -c clone.defaultRemoteName=origin
clone --bare --mirror https://github.com/openbmc/linux
/hdd/work/openbmc/build/romulus/downloads/git2/github.com.openbmc.linux
--progress failed with exit code 128, see logfile for output
ERROR: linux-aspeed-6.6.71+git-r0 do_fetch: Bitbake Fetcher Error:
FetchError('Unable to fetch URL from any source.', 'git://
github.com/openbmc/linux;protocol=https;branch=dev-6.6')
ERROR: Logfile of failure stored in:
/hdd/work/openbmc/build/romulus/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/6.6.71+git/temp/log.do_fetch.574827
ERROR: Task
(/hdd/work/openbmc/meta-aspeed/recipes-kernel/linux/linux-aspeed_git.bb:do_fetch)
failed with exit code '1
==================================================================

Then I found the link "gotjib.com/openbmc/linux" is valid and tried the URI
with "git clone" and it worked, but I'm not sure why bitbake could not
fetch the URI. other URI  with same format under the "poky" worked OK

So I tried the ssh uri from github.com/openbmc/linx and replaced that one
from the linux-aspeed and got bitbake failure.

So I guessed the problem from the URI, sorry for I missed one step to
verify if other worked URIs in same format with this one.  I should be more
careful.

Now I am confused, what's the root cause?

Thanks,

amboar, 2025-01-29:
```
failed with exit code 128, see logfile for output
```
```
ERROR: Logfile of failure stored in:
/hdd/work/openbmc/build/romulus/tmp/work/romulus-openbmc-linux-gnueabi/linux-aspeed/6.6.71+git/temp/log.do_fetch.574827
```

What is the content of the log file?

amboar, 2025-01-29:
The log looks incomplete.

> Also in the attachment.

There is no attachment on the issue. Please interact with the issue through the web interface:

https://github.com/openbmc/linux/issues/233

Chances are high the download failure was due to network flakiness. Bitbake often surfaces these issues. You may want to retry the build after removing any changes you've made to the bitbake metadata.

Please also provide the commit ID of openbmc/openbmc that you're trying to build from.